XRobots/ReallyUsefulRobot #1: How to change the Teensy serial port for rosserial

nbaddorf, 2020-12-22:
Hi, I have been working with the RUR programs for a couple of days and they work great! I would like to wire my teensy 4.1 to a raspberry pi for serial, rather than using usb. I tried to follow the rosserial tutorials on changing the wired serial port, but it throws me lots of errors, I am guessing this is because of the modified library. Does anyone know how I could change the teensy serial port used? Thanks!

XRobots, 2020-12-22:
In here: https://github.com/XRobots/ReallyUsefulRobot/blob/main/ROS/rur/launch/rur_bringup.launch  you can change the port name from /dev/ttyACM0 to something else, maybe /dev/ttyUSB0 ?

You may also have to do: sudo chmod 777 /dev/ttyUSB0 to give access.

nbaddorf, 2020-12-22:
Sorry, I guess I didn't explain well enough. I would like to change how the teensy is wired to the raspberry pi. I don't want to use the teensy over usb cable. I would like to wire the teensy to the rpi over serial2.

XRobots, 2020-12-22:
I guess you'd have to hack the rosserial library to use a different serial port.

nbaddorf, 2020-12-22:
I did try to use that, and rosserial actually has a tutorial on how to do that, which I tried, but it gives me errors, due to, I'm guessing, the mods you did the library. Here is the link to that tutorial: http://wiki.ros.org/rosserial_arduino/Tutorials/NodeHandle%20and%20ArduinoHardware

XRobots, 2020-12-22:
Changes I made were to https://github.com/ros-drivers/rosserial/blob/melodic-devel/rosserial_arduino/src/ros_lib/ArduinoHardware.h  to add "defined(__IMXRT1062__)" with the other hardware.

I also modified the buffer size in ros.h as per : https://answers.ros.org/question/73627/how-to-increase-rosserial-buffer-size/

nbaddorf, 2020-12-29:
Ok, thanks! I guess this bug is part of the rosserial library then, not due to the teensy stuff. Thanks for your help, I will keep you posted when I figure out the answer! By the way, I am a huge fan so this is super cool getting to ask for help!