XRobots/ReallyUsefulRobot #2: Running navigation stack and Gmapping in On-board raspberry Pi

Zeonsilt06, 2020-12-24:
Hi james. Can i ask you something about this project. Do you run all the Gmapping node and Navigation stack all on Raspberry Pi onboard or some of them in remote computer with SSH ? i also built the same system like you now, but can't figure out why the map update keep missing it's desired rate. I suspect some computation issue, but seeing your video with the same setting as me, it looks all good.

XRobots, 2020-12-24:
No, it's running on a remote workstation. I get that as well, apparently there's a known bug somewhere.

Zeonsilt06, 2020-12-24:
oh so you do the Gmapping Slam and navigation stack in remote workstation , but all serial node and rplidar node is in raspberry pi. Thank you very much !

XRobots, 2020-12-24:
Yep. I did put some notes in the readme about what runs where.

Zeonsilt06, 2020-12-24:
do you have performance issue running node through SSH, like Lidar data delay ?

XRobots, 2020-12-24:
Not that I know of - it seems to work ok