XRobots/ReallyUsefulRobot #5: TF error

chrismalyon, 2021-01-06:
Hi James,
  Im really struggling with one last bit. When i launch the rur_navigation package on my remote workstation i get two errors "no transform from Laser to Map and no transform from Base_link to Map. Im really stuck, can you point me in the right direction?

Thanks
Chris

XRobots, 2021-01-06:
Presumably you ran Gmapping and made the map ok?

chrismalyon, 2021-01-06:
No as I get a No Map received error but im currently using your map. I have tried to build this robot a few times but havent been able to. I was hoping I could do it now based on your videos but it seems im still struggling :(

XRobots, 2021-01-06:
You'll have to step back a bit. Just run the stuff on the robot and open RVIZ - can you add in the laser scan and TF and have it all line up?  After that you should be able to run Gmapping and have it draw the map.

chrismalyon, 2021-01-06:
Part of the problem i think is finding the correct distro to run. Im running a Raspberry Pi 4 and no matter what distro I use there always seems to be issues with getting ROS running properly. Right now for instance when I run your code I get an error with the serial connection to my Arduino due to a python version issue.

chrismalyon, 2021-01-06:
Ok I figured it out and I now have the robot communicating with the remote workstation and displaying the map correctly. The last bit is driving the robot. if i echo the odom topic and manually turn the wheels im getting the encoder count so that looks good however im not using ODrive im actually using a pair of hoverboard wheels with two BLDC drivers. these are driven essentially by a pwm signal. Im having trouble converting your code to make the wheels move. Below is what i have updated in the main loop. My thinking is to take the forward0 + turn0 and convert this to a PWM value however nothing happens :( I know im taking the micky a bit asking but if you could take a look at the below and let me know what im doing wrong i would appreciate it. I think others would find it useful to see how to make the code work in this way as pwm driven controllers and wheels are a lot cheaper than an ODrive setup.

Thanks in advance
Chris

  nh.spinOnce();        // make sure we listen for ROS messages and activate the callback if there is one

  currentMillis = millis();
  if (currentMillis - previousMillis >= loopTime) {  // run a loop every 10ms
    previousMillis = currentMillis;          // reset the clock to time it

        float modifier_lin = 1.03;        // scaling factor because the wheels are squashy / there is wheel slip etc.
    float modifier_ang = 0.92;        // scaling factor because the wheels are squashy / there is wheel slip etc.

    forward0 = demandx * (83466 * modifier_lin) ; // convert m/s into counts/s
    forward1 = demandx * (83466 * modifier_lin); // convert m/s into counts/s

    turn0 = demandz * (15091 * modifier_ang);    // convert rads/s into counts/s
    turn1 = demandz * (15091 * modifier_ang);    // convert rads/s into counts/s

    //forward1 = forward1 * -1;    // one motor and encoder is mounted facing the other way

//    odrive1.SetVelocity(0, forward0 + turn0);
//    odrive1.SetVelocity(1, forward1 + turn1);

      analogWrite(Left_Wheel_speed, forward0 + turn0);
      digitalWrite(Left_Wheel_Direction, HIGH);
      digitalWrite(Left_Wheel_Stop, HIGH);

      analogWrite(Right_Wheel_speed, forward1 + turn1);
      digitalWrite(Right_Wheel_Direction, LOW);
      digitalWrite(Right_Wheel_Stop, HIGH);
      

    // get positions and velocities from ODrive

//    pos0 = (odrive1.GetPosition(1)) * -1;
//    pos1 = odrive1.GetPosition(0);

    pos0 = pos;
    pos1 = pos2;

XRobots, 2021-01-07:
Well it looks ok, if you try the motor driver code on its own does it drive the motors?

chrismalyon, 2021-01-07:
It does. I have done some debugging via Serial1 on the arduino mega. It seems that the Arduino is not receiving the cmd_vel topic so there seems to be some issue with how it is subscribing. It is however publishing ok as when i rotate the wheels manually the encoder count is updated to the odom topic which i can see with rostopic echo /odom on the remote workstation. I will continue playing with it to see if I can work out whats going on.

chrismalyon, 2021-01-07:
Hmmm I thought it could be an issue with the Arduino Libraries i was using so I recompiled but didnt work. I have added the following to the callback function on the Arduino but It doesnt seem to fire. If I rostopic echo cmd_vel on the raspberry on the robot then i can see the topic getting the data from rviz but the arduino doesnt seem to capture it :'(

void velCallback(  const geometry_msgs::Twist& vel)
{
  
  demandx = vel.linear.x;
  demandz = vel.angular.z;

  Serial1.print("Callback X: ");
  Serial1.println(vel.linear.x);

  Serial1.print("Callback Z: ");
  Serial1.println(vel.linear.z);
}

matthewjreeves, 2021-02-09:
Hi chrismalyon 

> Ok I figured it out and I now have the robot communicating with the remote workstation and displaying the map correctly. 

Earlier in this thread, you had exactly the same problem (above) as I am currently having. Running Rviz after launching all the nodes is throwing up the error Transform [sender=unknown_publisher]
For frame [laser]: No transform to fixed frame [map]. TF error: [Could not find a connection between 'map' and 'laser' because they are not part of the same tree.Tf has two or more unconnected trees.]

I am really stuck too! Could you shed some light on how you fixed it. I am also trying to run it all on an RPI 4b and have had a hell of a time trying to compile Realsense (which is now sorted thankfully)!