XRobots/ReallyUsefulRobot #8: the robot moves faster in rviz then in real

bilx6043, 2021-04-19:
hi... thanks for all the hardwork you put on this project and make it working.. i try your code with my robot using audriuino mega..but my robot runs faster in rviz then in real .. and i think cuz of that the navigation stack also not work..i have tried the wheel reduction ratio part and calculate the values according to my robot but still no luck..

XRobots, 2021-04-19:
The ODrive firmware has been updated since I built this to use 'turns' instead of encoder counts as the units. So the amount the wheels turn will need rescaling.

bilx6043, 2021-04-20:
thanks for the reply... so that is the issue... i am using your audrino code to connect to odrive... i have calculated the wheel revolution and put it on the code... i have calculated check revolution per second with speed and its ok.. not when i send speed from keyboard, lets say 0.5 .. on the odom topic i got different speed its 9.5 .. so i think its cuz of the odive firmware update causing the issue or i am missing something?

abdullahkose20, 2021-04-20:
Hi.. bruton
 ı have studied on ros and odrive for around 6 mounths. I appreciate your studies. Previously, we didnt have anyproblem about controlling odrive and ros but now, due to odrive update firmware, we faced same problem. My robot worked stable before, now it is unstable and so ı can not reach my targets. Can u help me with update version of odrive. thanks for your answer

XRobots, 2021-04-20:
Yep here's the ODrive migration guide to the new firmware: https://docs.odriverobotics.com/migration