XRobots/ReallyUsefulRobot #17: Question - faisability and estimated replication time

thibnoel, 2021-11-03:
Hello,

I'm currently pursuing a PhD in robotics, specifically about autonomous exploration and reconstruction of an unknown environment. We are currently using another platform for the project but it is not so reliable mechanically, and seeing the design of the RUR, it seems like it would be a great alternative for us, especially given we use very similar sensors (same LiDAR and a Kinect for RGBD vision). We would probably use an arm-less version since we don't plan any manipulation tasks yet.

I am currently discussing with my supervisors the possibility to try and replicate the RUR during my PhD but it is a bit difficult for me to estimate the time it would take for a team of 1-3 people to rebuild it; we have pretty good knowledge in the team about 3D printing and mechanical assembly so this should not be an issue, it's more on the electronics side that I'm not so sure about the difficulty of such an assembly. This leads me to the following questions : 
- how long do you think it would approximately take to wire, test and integrate the wheeled-base electronics with the mechanical assembly for a team of 1-3 people with a beginner/intermediate level in electronics ?
- do you see any steps in the building process that we would need to pay special attention to ?
- if we end up rebuilding and using the RUR for a research project, how would you prefer to be credited (YouTube channel, website, you directly) ?

Thanks in advance for any feedback about this and for the great inspiration your work is in general !
Thibault

XRobots, 2021-11-03:
It took me about 2 weeks on my own to build it all and get the base up and running and navigating (the first two videos in the playlist). I do have a lot of printers though so I could make all the parts concurrently. I was still learning about ROS at the same time and this is the first ROS robot I build that worked properly.

https://www.youtube.com/playlist?list=PLpwJoq86vov_wIh6f-A1ZkjCwo11jkbe3

One issue now is that ODrive have updated their firmware and Arduino Library, so all the units are now different (turns instead of encoder counts), so you'll have to work through converting the units if you buy a new ODrive.

Feel free to credit my YouTube if you wish, although the whole thing is open source, so just publishing any modified source would be fine.

thibnoel, 2021-11-03:
Thanks a lot for your answer ! Do you know if anyone has attempted/succeeded to replicate the project yet ?
I guess the new ODrive firmware should not be too much of an issue if it's only a matter of unit conversions

XRobots, 2021-11-03:
I'm not aware of anyone who's completed it. A few others have asked question - I believe there are some closed issues.