XRobots/ReallyUsefulRobot #18: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100065 timeout was 0.1

MiPlayer123, 2022-01-05:
Hello

I am running into issues when running `roslaunch rur_navigation rur_navigation.launch` with `roslaunch rur rur_bringup.launch`. 
This is the error:
```
process[map_server-1]: started with pid [27581]
process[amcl-2]: started with pid [27582]
process[move_base-3]: started with pid [27583]
process[rviz-4]: started with pid [27596]
[ WARN] [1641341048.747458020]: Request for map failed; trying again...
[ WARN] [1641341053.709191464]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100065 timeout was 0.1.
[ WARN] [1641341058.731377300]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1003 timeout was 0.1.
[ WARN] [1641341063.771360315]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100305 timeout was 0.1.
[ WARN] [1641341065.510832061]: No laser scan received (and thus no pose updates have been published) for 1641341065.510745 seconds.  Verify that data is being published on the /scan topic.
[ WARN] [1641341068.811848131]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.100372 timeout was 0.1.
```

I am not sure where the error is coming from. Thanks in advance.

XRobots, 2022-01-05:
Looks like the map is missing as well as the laser data. Can you actually see the laser data getting published on /scan in RVIZ and does it sync up with TF?

Have you actually created the map yet?

MiPlayer123, 2022-01-06:
For now, I was just using the default map, which is set as `house_map` in the navigation launch file `<arg name="map_file" default="$(find rur_navigation)/maps/house_map.yaml"/>`. I am not sure why the error exists as the map and pgm file exist.

I use `rur_bringup.launch`, which has the laser scan included `<include file="$(find rplidar_ros)/launch/rplidar.launch" />`, but I don't see any data being published in RVIZ. When I run the rplidar node separately, I am able to see the data being published.

3087575358, 2023-06-07:
I got the same question, which shows that the status of laserscan and others are error,have you get done with that yet?
[image]

3087575358, 2023-06-07:
I got the same question, which shows that the status of laserscan and others are error,have you got done with that yet?
[image]