XRobots/ReallyUsefulRobot #19: Robot vibrating

LinusNEP, 2022-02-07:
Hello,
The simulation worked fine for me, although the robot always gets stuck on a narrow passage during autonomous navigation. However, I am trying to implement this on the real robot but no luck at the moment. 
I have an ODrive controller 56v v3.6, Arduino mega and jetson nano but a different wheel (hoverboard 165.1 mm wheel diameter). 
When I run the Arduino ROS serial node, the wheel spins but with heavy vibration. I anticipate it could be errors due to the encoder and wheel parameters calculation. Also, the robot moves quite slow in RViz (sometimes unnoticeable) compared to real-time.
I am wondering if you could spare some time to assist me to fix it.

XRobots, 2022-02-08:
ODrive updated their firmware to use 'turns', which is a motor revolution, instead of encoder counts. So you'll probably need to adjust all the units because they are 8192 times too big now.

LinusNEP, 2022-02-08:
@XRobots Thanks for you reply. So in this case,  I have to divide the previous  values by 90 cpr to get it in turns/m, turns/s and rad/s (the values written in red). I am wondering if that is right? 
[image]

XRobots, 2022-02-09:
It looks like the right process, but I haven't checked the maths ;-)

LinusNEP, 2022-02-09:
Ok, thank you very much for your concern. I would be pleased if you could find out time to go through it. I tested the new values with the robot and it seems to work fine at low speed. However, there are still vibration driving the robot at high speed.