XRobots/ReallyUsefulRobot #20: GR-WM2 replacement?

kssummer, 2022-02-10:
Is it possible to source the GR-WM2 somewhere? Or is there a suitable replacement?

Looks like the carriage motor has been discontinued: https://gimsonrobotics.co.uk/categories/dc-electric-motors/products/gr-wm2-24v-dc-rs-595-motor-with-152-1-gearbox?fbclid=IwAR0d2dX7y88nv4W90gfV4aULetfhDwCAWDCMk14pTz_aJ7J1FpBhlCw4sK0

They have a replacement, but it has some issues:
(a) the motor (gearbox) mounting holes are only on one side, which might not allow for enough support from the 3D printed mounts
(b) I expect the shaft won't be lined up the same as the GR-WM2. The belt needs to be driven vertically to move the carriage, which means we'd have to redesign the CAD to line up the motor, idler, and carriage.

Issue (a) isn't that big of a deal, I could fab a piece of aluminum/steel to securely mount the motor, but issue (b) is a big deal for me. It requires some CAD redesign I don't have the chops for. Also, I expect any replacement will have this same issue unless the shaft offset is exactly the same.

XRobots, 2022-02-11:
I bought a bunch of those motors when I built Performance Robots so I've not really investigated a replacement I'm afraid. I guess it'll take some redesign to incorporate a totally different motor.

kssummer, 2022-02-11:
Hi James,

Any chance you'd be willing to sell me one of those motors?

XRobots, 2022-02-11:
Sorry I don't have enough spare - there are eight on robots at the moment and I have two extras which I use when I need powerful differential drive in a project, but they are also the two spares for Performance Robots if I take them to events again.

kssummer, 2022-02-11:
No worries. Thanks for the response and thanks for all your work on these
cool robot projects