XRobots/ReallyUsefulRobot #22: Wiring diagram?

kssummer, 2022-02-20:
Is there a wiring diagram for this robot?
I'm trying to understand how the relay, switches, voltmeters, and control boards are connected.

XRobots, 2022-02-20:
There isn't a wiring diagram, but the details are here: https://youtu.be/dEPvZZ4luV4?list=PLpwJoq86vov_wIh6f-A1ZkjCwo11jkbe3&t=202

The switches turn on power from the two batteries, the 12V one is for the Xavier and the 24V one is for the ODrive. The meters are wired after the switches so they only come on when the power is on. The relay is used to switch on the power to the ODrive because the 10A toggle isn't enough to switch 20A per axis.

filoucool, 2022-04-14:
Hi! I also had a question about the wiring between the Odrive and the Teensy... To what Pins on the Odrive and teensy do the white and yellow cables go to? I think I figured out that it goes to the teensy's Serial1 pins but I cant figure out for the Odrive...
Thanks!

XRobots, 2022-04-14:
You need to look at the ODrive documentation so your wiring matches the current firmware: https://docs.odriverobotics.com/v/latest/pinout.html

closed 2022-04-14 by XRobots