XRobots/ReallyUsefulRobot #26: Navigation failure

kssummer, 2022-10-05:
I'm having trouble with the navigation stack. When I make a waypoint, my robot starts jerking around; no smooth movement. Today the robot fell over because of this. Any idea on what could cause this? Or how to fix it?

For background, I was able to get the robot moving smoothly. I had to change the arduino code to convert the Odrive turns to millimeters, but now its fine. The odom, base_link, and lazer show up in rviz. I can drive around fine, the lazer scan stays put, and I have built a map of my living room. I launched the navigation stack, the map comes up, although the costmap looks rather inflated, but the robot is oriented right, assuming I place it in my room where I started the map.

It looks like the navigation stack is continuously updating its plan. The robot never really gets a chance to drive straight; instdead it whips back and forth like a low-rent murder bot. My dogs are concerned.

(I rolled back the git repository to the original navigation commit, before the simulation stuff was merged just to be sure that wasnt the issue, but both versions have this issue. In the latest code, the lazer scan gets out of synch with the map; that doesn't happen in the original version)

Here's the log from the navigation:
roslaunch rur_navigation rur_navigation.launch map_file:=$HOME/ros_catkin_ws/src/rur_navigation/maps/livingroom_sean.yaml
... logging to /home/sean/.ros/log/546ce954-4473-11ed-920f-bfb2a73f92b8/roslaunch-beefcake-62996.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://beefcake:38965/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_link
 * /amcl/gui_publish_rate: 50.0
 * /amcl/initial_pose_a: 0.0
 * /amcl/initial_pose_x: 0.0
 * /amcl/initial_pose_y: 0.0
 * /amcl/kld_err: 0.02
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 180
 * /amcl/laser_max_range: 3.5
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 3000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.1
 * /amcl/odom_alpha2: 0.1
 * /amcl/odom_alpha3: 0.1
 * /amcl/odom_alpha4: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 0.5
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.2
 * /move_base/DWAPlannerROS/acc_lim_theta: 3.2
 * /move_base/DWAPlannerROS/acc_lim_x: 2.5
 * /move_base/DWAPlannerROS/acc_lim_y: 0.0
 * /move_base/DWAPlannerROS/controller_frequency: 10.0
 * /move_base/DWAPlannerROS/forward_point_distance: 0.325
 * /move_base/DWAPlannerROS/goal_distance_bias: 20.0
 * /move_base/DWAPlannerROS/latch_xy_goal_tolerance: False
 * /move_base/DWAPlannerROS/max_scaling_factor: 0.2
 * /move_base/DWAPlannerROS/max_vel_theta: 2.75
 * /move_base/DWAPlannerROS/max_vel_trans: 0.22
 * /move_base/DWAPlannerROS/max_vel_x: 0.22
 * /move_base/DWAPlannerROS/max_vel_y: 0.0
 * /move_base/DWAPlannerROS/min_vel_theta: 1.37
 * /move_base/DWAPlannerROS/min_vel_trans: 0.11
 * /move_base/DWAPlannerROS/min_vel_x: -0.22
 * /move_base/DWAPlannerROS/min_vel_y: 0.0
 * /move_base/DWAPlannerROS/occdist_scale: 0.02
 * /move_base/DWAPlannerROS/oscillation_reset_dist: 0.05
 * /move_base/DWAPlannerROS/path_distance_bias: 32.0
[... 125 lines cut ...]
[ INFO] [1664949841.710613075]: Got new plan
[ INFO] [1664949841.910624391]: Got new plan
[ INFO] [1664949842.110641782]: Got new plan
[ INFO] [1664949842.310630494]: Got new plan
[ INFO] [1664949842.510672744]: Got new plan
[ INFO] [1664949842.710608765]: Got new plan
[ INFO] [1664949842.910698871]: Got new plan
[ INFO] [1664949843.110593497]: Got new plan
[ INFO] [1664949843.310540360]: Got new plan
[ INFO] [1664949843.510582199]: Got new plan
[ INFO] [1664949843.710666019]: Got new plan
[ INFO] [1664949843.910593931]: Got new plan
[ INFO] [1664949844.110599047]: Got new plan
[ INFO] [1664949844.310571304]: Got new plan
[ INFO] [1664949844.510679543]: Got new plan
[ INFO] [1664949844.710543665]: Got new plan
[ INFO] [1664949844.910458401]: Got new plan
[ INFO] [1664949845.110632632]: Got new plan
[ INFO] [1664949845.310576916]: Got new plan
[ INFO] [1664949845.510588454]: Got new plan
[ INFO] [1664949845.710548413]: Got new plan
[ INFO] [1664949845.910618182]: Got new plan
[ INFO] [1664949846.110646003]: Got new plan
[ INFO] [1664949846.310541918]: Got new plan
[ INFO] [1664949846.510692447]: Got new plan

XRobots, 2022-10-05:
Hello, I don't know what to do with this, I haven't looked at the robot for years.

ODrive updated their firmware so the units are different. I have no easy way to test your pull request before merging it. Feel free to fork the repository / publish your own version though.

kssummer, 2022-10-05:
@PranshuTople maybe you can help point me in the right direction here? I updated the arduino code to convert the new units from the new ODrive firmware. I can drive the robot around with the rosremote and create a good map, but whenever I make a way point with the navigation stack, the robot stutters its movement. Any ideas on what I can look at?

PranshuTople, 2022-11-17:
Hey @kssummer,

Can you share a video of your robot's motion first in teleoperation mode and later in navigation? I want to see how it behaves in order to comment on anything

FrancisTembo, 2023-07-23:
Couldn't get the unit conversion right. @kssummer would you mind sharing your fix?